jdeath/rd200v2 #1: Issue with Adding Manually Through Integrations Page

Natrinicle, 2022-12-25:
When attempting to add a device manually via the integrations page it tells me there was a 500 error and this stack trace appears in the logs.

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 222, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 81, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 136, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 148, in post
    return await super().post(request)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 73, in wrapper
    result = await method(view, request, data, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 71, in post
    result = await self._flow_mgr.async_init(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 226, in async_init
    flow, result = await task
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 253, in _async_init
    result = await self._async_handle_step(flow, flow.init_step, data, init_done)
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 368, in _async_handle_step
    result: FlowResult = await getattr(flow, method)(user_input)
  File "/config/custom_components/rd200_ble/config_flow.py", line 146, in async_step_user
    if discovery_info.advertisement.local_name is not None or not discovery_info.advertisement.local_name.startswith("FR:RU"):
AttributeError: 'NoneType' object has no attribute 'startswith'
```

jdeath, 2022-12-25:
Darn, had some bad logic. Try the latest version of config_flow.py

Natrinicle, 2022-12-25:
Looks like it's no longer throwing the 500 but this time it's showing

```
Please wait, starting configuration wizard for RD200 BLE
```
and then appears to time out while attempting to configure the device
```
Config flow could not be loaded: <html><body><h1>504 Gateway Time-out</h1> The server didn't respond in time. </body></html>
```

Natrinicle, 2022-12-26:
When I bypass the web proxy in front of my HA instance I see this instead of gateway timeout.

```
Failed to connect
```

and the logs show

```
Error connecting to and getting data from 24:D7:EB:5E:B6:8A: 24:D7:EB:5E:B6:8A - 24:D7:EB:5E:B6:8A: Failed to connect: 
```

The official application is able to connect though I don't have it connected while I'm attempting to connect with this integration to make sure they don't collide.

jdeath, 2022-12-26:
I found the connection to be touchy. Move it a little closer to the HA or ESPHome gateway. Also, make sure nothing else is connected. If you have a bluetooth plugged into a USB port, but it on a port extender.

I had my ESPHome connected as a BLEClient and it would not connect. Took a while to figure that out.
If you turn on debugging, you'll get some more logs.
```
logger:
  default: error
  logs:
    custom_components.rd200_ble: debug
```

Natrinicle, 2022-12-26:
Because the ESPHome component didn't support v2 of the device I only had it connecting for a few hours when I first received the RD200 and the only ESP32 running the Bluetooth Proxy is ~20 feet away through a floor. I've only got the official app on a single phone and I force closed the application to make sure it wasn't connected. I added the custom_components logger level at debug and retried running through the setup with the RD200 sitting less than a foot from the Bluetooth adapter. Here are the relevant logs

```
2022-12-26 10:13:21.367 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] Found My Device
2022-12-26 10:13:21.367 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] RD2000 Discovery address: 24:D7:EB:5E:B6:8A
2022-12-26 10:13:21.368 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] RD2000 Man Data: {}
2022-12-26 10:13:21.368 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] RD2000 advertisement: AdvertisementData(local_name='FR:RU22201030184', service_uuids=['0000abf0-0000-1000-8000-00805f9b34fb'], rssi=-46)
2022-12-26 10:13:21.370 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] RD2000 device: 24:D7:EB:5E:B6:8A: FR:RU22201030184
2022-12-26 10:13:21.370 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] RD2000 service data: {}
2022-12-26 10:13:21.370 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] RD2000 service uuids: ['0000abf0-0000-1000-8000-00805f9b34fb']
2022-12-26 10:13:21.370 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] RD2000 rssi: -46
2022-12-26 10:13:21.370 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] RD2000 advertisement: FR:RU22201030184
2022-12-26 10:14:42.397 ERROR (MainThread) [custom_components.rd200_ble.config_flow] Error connecting to and getting data from 24:D7:EB:5E:B6:8A: 24:D7:EB:5E:B6:8A - 24:D7:EB:5E:B6:8A: Failed to connect:
```

Natrinicle, 2022-12-26:
The logging output shows a different service UUID than what's in your parser but looking in BLE-Monitor on my Android I definitely see both the read and write characteristic UUIDs under a service with the UUID 00001523-0000-1000-8000-00805f9b34fb so it looks like my RD200 is advertising the correct UUIDs, it's just that HomeAssistant that can't find them for some reason...

jdeath, 2022-12-26:
This is weird. It found your device, but will not connect. I think the UUIDs are fine, doubt you have a special model.
Are you sure you do not the MAC address still in the esphome ble_client: line? I did, and it kept me from connecting. Do you have your proxy set as active(not sure if matters)?

```
bluetooth_proxy:
  active: true
```

Natrinicle, 2022-12-26:
My whole ble_client section was commented out after I realized that it wasn't going to work with my version. I've flashed that device several times since then thanks to the bluetooth_proxy updates. I am also able to connect with my phone if I use the app so I really don't think anything else is connected. Here are the relevant sections from my ESPHome though just in case.

```yaml
# ble_client:
#   - mac_address: 24:D7:EB:5E:B6:8A
#     id: radon_eye_ble_id
```

```yaml
bluetooth_proxy:
  active: true
esp32_ble_tracker:
  scan_parameters:
    interval: 320ms
    window: 320ms
```

jdeath, 2022-12-26:
Maybe remove the interval and window? I think that makes it scan all the time, so little time to write. 
Mine looks like:
```
esp32_ble_tracker:
  scan_parameters:
    active: true

bluetooth_proxy:
  active: true
```

Natrinicle, 2022-12-26:
I reflashed with these settings

```yaml
bluetooth_proxy:
  active: true
esp32_ble_tracker:
  scan_parameters:
    active: true
    # interval: 320ms
    # window: 320ms
```

and got the same results in HomeAssistant but the ESPHome logs were still open in another browser and I saw

```
[11:14:12][I][bluetooth_proxy:254]: [0] [0B:17:DD:89:A0:64] Connecting v3 without cache
[11:14:12][D][esp32_ble_tracker:216]: Pausing scan to make connection...
[11:14:12][I][esp32_ble_client:064]: [0] [0B:17:DD:89:A0:64] 0x01 Attempting BLE connection
[11:14:14][D][sensor:127]: 'BLE Receiver Free Heap': Sending state 80492.00000 B with 1 decimals of accuracy
[11:14:22][I][esp32_ble_client:078]: [0] [0B:17:DD:89:A0:64] Disconnecting.
[11:14:26][D][sensor:127]: 'BLE Receiver Min Free Heap': Sending state 31164.00000 B with 1 decimals of accuracy
[11:14:34][D][sensor:127]: 'BLE Receiver Free Heap': Sending state 80492.00000 B with 1 decimals of accuracy
[11:14:42][W][esp32_ble_client:131]: [0] [] Connection failed, status=133
```

I unplugged my only ESPHome with a bluetooth_proxy enabled so that only the USB dongle that was ~6" away was available and tried again but with the same dialogs and logs.

jdeath, 2022-12-26:
Not sure what those logs are (I am not a BT expert, but that is not your RD200 MAC).

Maybe power cycle the RD200? Not sure what else can be done. I just removed my integration and restarted. It found it when I manually added it. I do switch between app, python script and integration.

Since you fail the "connect" step, it does not even get to where it reads the characteristic.

You are the only one I have heard from, so maybe there is something cached in my setup that allows it to connect.

Perhaps try manually adding it a few times in a row and maybe it will connect? Sorry it is giving you trouble.

Natrinicle, 2022-12-27:
> Not sure what those logs are (I am not a BT expert, but that is not your RD200 MAC).

It's from the ESPHome logging output (the stuff that comes up right after you flash a device again through the web GUI.

I ended up rebooting the server my HomeAssistant is running on and after it came back up, HomeAssistant automatically detected the RD200 and offered to let me configure it. I hit that and everything worked perfectly... Not sure what was going on but it seems like an issue that was fixed by: "When in doubt, reboot it out!" Thanks for all the help and troubleshooting.

closed 2022-12-27 by Natrinicle