jdeath/rd200v2 #4: local_name filtering leaves out my device

arpia49, 2022-12-29:
Hello, in the manifest the local_name looks like:

`
  "bluetooth": [
    {
      "local_name": "FR:RU*"
    }
  ]
`

But my device is _FR:RE_, I wonder if the E stands for España (Spain). If I set up in like FR:RE, the integration works, if I don't make the change, it does not recognize my device.

jdeath, 2022-12-29:
Yes, the docs said it only works for FR:RU. Didn't know they made another one.

I surprised the manifest did anything. Did you need to change the config_flow.py? There is another filter there on the serial number? I think I can add another entry for the RE version

jdeath, 2022-12-29:
Closed with version 1.2

closed 2022-12-29 by jdeath

arpia49, 2022-12-29:
That was quick! I didn't check the config_flow.py, but maybe that was the reason of some issues I had... let me play with the new version and I will confirm if everything is working as expected.