jdeath/rd200v2 #6: serial number starting with RE2220

jonathan-schmieg, 2023-01-06:
Hello jdeath!

I appreciate the work you did on integrating the RD200 with home assistant.
I have a question regarding the serial number:
the number printed on the device and displayed in the OLED starts with RE2220, however when I run a bluetoothctl scan I get
`[NEW] Device 24:4C:AB:03:20:A6 FR:RE22203281468`
which shows the FR:RE part.
Do you have "FR:" in your serial number as well?
When I try to add your integration I get
`No devices found on the network`
Any ideas what I might be missing?

thank you for your help,
Jonathan

jdeath, 2023-01-06:
The display does not show the FR, so yours is fine.

Make sure you do not have the MAC listed in esphome BLE client section, as nothing else should be connected to it.

Turn debugging on, to get some more information:
```
logger:
  default: error
  logs:
    custom_components.rd200_ble: debug
```

jonathan-schmieg, 2023-01-06:
I might have missed an important bit there (maybe you could add it to the readme):
Do I still need the esphome client as a bluetooth proxy?
I thought this integration talks directly with the RD200.

jdeath, 2023-01-06:
You need either an ESPHome proxy or a BT adapter on home assistant box (both is fine too). Sounds like you have BT adapter on you homeassistant, which is fine.

Many people first try the ESPHome component for the V1 (which does not work), and forget to remove the configuration in esphome.

jonathan-schmieg, 2023-01-06:
Thank you for the quick response!
I'm using a raspberry pi 4 with the home assistant image flashed on it. There's no ESP32 connected to the RD200 anymore.
Where can I add the debugging stanza?

jdeath, 2023-01-06:
Do you see the BT integration in your list? That needs to be setup. 

Those lines go in your homeassistant configuration.yaml. You may already have the: 
```
logger:
  default: error
```

jonathan-schmieg, 2023-01-06:
I see an integration
`E4:5F:01:6B:B3:0C
Bluetooth
1 device`
the only configuration option that I have in there is "passive scanning" which is not ticked.

I added the entire logger stanza to my configuration.yaml and restarted HA.

I don't see anything in the "UI-logs", using the file editor I only see

`2023-01-06 12:53:55.324 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-01-06 12:53:55.328 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration rd200_ble which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant`

trying to add the integration still results in the same error and does not add any new lines to the log

jdeath, 2023-01-06:
Ohh, crap. Something must have gotten messed up. I just pushed another release (1.4), try that. Somehow the line looking for the spain version got changed to look for the US version.

jonathan-schmieg, 2023-01-06:
I updated to the new version and there's progress!
I'm getting a different error message in the logs:

```
This error originated from a custom integration.

Logger: custom_components.rd200_ble.config_flow
Source: custom_components/rd200_ble/config_flow.py:165
Integration: RD200 BLE (documentation)
First occurred: 13:53:26 (4 occurrences)
Last logged: 14:02:56

Error connecting to and getting data from 24:4C:AB:03:20:A6: 24:4C:AB:03:20:A6 - 24:4C:AB:03:20:A6: Failed to connect: [org.bluez.Error.Failed] le-connection-abort-by-local: Interference/range; External Bluetooth adapter w/extension may help; Extension cables reduce USB 3 port interference```

Going to try an esp bluetooth proxy and report back once I got the time to test it.

Thank you very much for your help

jdeath, 2023-01-06:
Also try a cold reboot of your Pi. Sometimes that can help.

jonathan-schmieg, 2023-01-07:
Progress!
After I set up an esp bluetooth proxy I was able to initialize and connect to the RD200.
However it _only works on the first iteration_, then I have to restart home assistant and I get another reading - but also only one.
Here's the logs from last night's attempt:
```
2023-01-07 01:19:54.587 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7f92bc2890 [unset, waiters:1]>
2023-01-07 01:19:54.779 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7f8640bfa0 [unset, waiters:1]>
2023-01-07 01:19:54.784 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 14.278 seconds (success: True)
2023-01-07 01:19:54.789 DEBUG (MainThread) [custom_components.rd200_ble.sensor] got sensors: {'radon': 68.0, 'radon_1day_level': 0.0, 'radon_1month_level': 0.0, 'radon_peak': 72.0}
2023-01-07 01:20:44.583 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] Discovered BT device: <home_assistant_bluetooth.models.BluetoothServiceInfoBleak object at 0x7f938e24c0>
2023-01-07 01:25:04.583 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7f80586530 [unset, waiters:1]>
2023-01-07 01:25:04.792 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7f805d1f00 [unset, waiters:1]>
2023-01-07 01:25:04.797 ERROR (MainThread) [custom_components.rd200_ble] Error fetching rd200_ble data: Unable to fetch data: unpack requires a buffer of 2 bytes
2023-01-07 01:25:04.802 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 10.551 seconds (success: False)
2023-01-07 01:30:04.257 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 0.000 seconds (success: False)
2023-01-07 01:35:04.253 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 0.000 seconds (success: False)
2023-01-07 01:40:04.257 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 0.000 seconds (success: False)
2023-01-07 01:45:04.251 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 0.000 seconds (success: False)
2023-01-07 01:50:04.252 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 0.000 seconds (success: False)
2023-01-07 01:55:04.252 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 0.000 seconds (success: False)
2023-01-07 02:00:04.252 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 0.000 seconds (success: False)
2023-01-07 02:05:04.257 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 0.000 seconds (success: False)
...
goes on like this - every 5 minutes one line like that
```
restarting home assistant now
this is the new log:

```
2023-01-07 10:40:05.280 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7fa28496c0 [unset, waiters:1]>
2023-01-07 10:40:05.457 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7fb3c9d720 [unset, waiters:1]>
2023-01-07 10:40:05.461 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 24.426 seconds (success: True)
2023-01-07 10:40:05.468 DEBUG (MainThread) [custom_components.rd200_ble.sensor] got sensors: {'radon': 97.0, 'radon_1day_level': 0.0, 'radon_1month_level': 0.0, 'radon_peak': 120.0}
2023-01-07 10:40:44.531 DEBUG (MainThread) [custom_components.rd200_ble.config_flow] Discovered BT device: <home_assistant_bluetooth.models.BluetoothServiceInfoBleak object at 0x7fa34ea940>
2023-01-07 10:45:07.150 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7f906a2ec0 [unset, waiters:1]>
2023-01-07 10:45:07.537 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7f9f3f36d0 [unset, waiters:1]>
2023-01-07 10:45:07.539 ERROR (MainThread) [custom_components.rd200_ble] Error fetching rd200_ble data: Unable to fetch data: unpack requires a buffer of 2 bytes
2023-01-07 10:45:07.551 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 2.290 seconds (success: False)
```
One thing I noticed is that on the first connection the bluetooth symbol on the RD200 disappeared after transferring the data. On the second (and all upcoming) try the the bluetooth symbol stays on but with that error.

jdeath, 2023-01-07:
Hmm..It stays on? It is almost like it is not disconnecting. Thanks for noticing. That could be something!

jonathan-schmieg, 2023-01-07:
Awesome now it seems to be working fine!
I can confirm that with version 1.4 the connection was only terminated after the first try (which was successful) and later stayed open forever.
Now on version 1.5 the connection gets established, data is transferred and then it's closed again.
As I still have the debug log enabled I'd like to send you the output as well. There's still a KeyError: 'radon_peak' in there.  It feels like something you might want to catch.
The radon peak gets reported to home assistant and matches the value on the RD200 display though.

Here's two samples of the logfile - each block repeats after 5 minutes but I get the radon level in home assistant which makes me really happy!
I am very grateful for your work thank you so much!

```
2023-01-07 19:17:45.614 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 0
2023-01-07 19:17:45.667 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 1
2023-01-07 19:17:47.154 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7f762e6200 [unset, waiters:1]>
2023-01-07 19:17:47.155 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 2
2023-01-07 19:17:47.159 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 3
2023-01-07 19:17:47.159 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] bytearray(b'P\nQ\x00\x00\x00\x00\x00\x02\x00\x07\x00')
2023-01-07 19:17:47.159 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 4
2023-01-07 19:17:47.159 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 5
2023-01-07 19:17:47.199 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 5.5
2023-01-07 19:17:47.562 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7f7625a6b0 [unset, waiters:1]>
2023-01-07 19:17:47.563 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 6
2023-01-07 19:17:47.567 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 7
2023-01-07 19:17:47.567 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] bytearray(b'@B220328RE21468\x06RD20')
2023-01-07 19:17:47.568 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 8
2023-01-07 19:17:47.568 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] Try to disconnect
2023-01-07 19:17:47.569 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] disconnect
2023-01-07 19:17:47.570 DEBUG (MainThread) [custom_components.rd200_ble] Finished fetching rd200_ble data in 2.239 seconds (success: True)
2023-01-07 19:17:47.571 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 168, in _handle_refresh_interval
    await self._async_refresh(log_failures=True, scheduled=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 316, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 121, in async_update_listeners
    update_callback()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 381, in _handle_coordinator_update
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 556, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 597, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 562, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 855, in state
    value = self.native_value
  File "/config/custom_components/rd200_ble/sensor.py", line 165, in native_value
    return self.coordinator.data.sensors[self.entity_description.key]
KeyError: 'radon_peak'
2023-01-07 19:22:47.888 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 0
2023-01-07 19:22:47.942 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 1
2023-01-07 19:22:49.552 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7f75dd4a30 [unset, waiters:1]>
2023-01-07 19:22:49.554 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 2
2023-01-07 19:22:49.556 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 3
2023-01-07 19:22:49.556 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] bytearray(b'P\nQ\x00\x00\x00\x00\x00\x06\x00\x07\x00')
2023-01-07 19:22:49.556 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 4
2023-01-07 19:22:49.556 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 5
2023-01-07 19:22:49.650 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 5.5
2023-01-07 19:22:49.952 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] self._event <asyncio.locks.Event object at 0x7f8a9a58d0 [unset, waiters:1]>
2023-01-07 19:22:49.953 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 6
2023-01-07 19:22:49.956 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 7
[... cut ...]
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 597, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 562, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 855, in state
    value = self.native_value
  File "/config/custom_components/rd200_ble/sensor.py", line 165, in native_value
    return self.coordinator.data.sensors[self.entity_description.key]
KeyError: 'radon_peak'
```

jdeath, 2023-01-07:
Awesome! I'll look into that key error. I do not have it.

Ahh, i see the issue and why you are having a problem (and the connection issues this whole time). When asking for the peak radon, your device is not returning it. It only returns about 1/3 the data:
```
DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] 7
2023-01-07 19:22:49.957 DEBUG (MainThread) [custom_components.rd200_ble.rd200_ble.parser] bytearray(b'@B220328RE21468\x06RD20')
```
Therefore, the peak key is never filled, because I now check if that is the correct 68 bytes long. I guess I could try/catch that, or fill it with None value or something. Let's see if your device ever reports it.

closed 2023-01-07 by jdeath

jonathan-schmieg, 2023-01-09:
@jdeath the weird thing is: the peak value was reported all the time - just every 5 minutes I get this exception in the log.
I modified `custom_components/rd200_ble/sensor.py` locally and implemented your suggestion:
```
    @property
    def native_value(self) -> StateType:
        """Return the value reported by the sensor."""
        try:
            return self.coordinator.data.sensors[self.entity_description.key]
        except KeyError:
            return None
```
Everything is still working fine but the logs are clean now.
If you don't mind I would appreciate if you could add this to the code so I'm able to follow potential updates you might do. I can imagine that I'm not the only one with this issue.

Thanks again for your work!

arpia49, 2023-01-09:
> I updated to the new version and there's progress! I'm getting a different error message in the logs:
>
> ```
> This error originated from a custom integration.
>
> Logger: [...]

Make sure you are not using the USB3 ports... (the blue ones), they make a ton of interfereces.

jdeath, 2023-01-09:
@jonathan-schmieg 

[bef1137](https://github.com/jdeath/rd200v2/commit/bef1137b3e12c1933e494ce44c0c236957082db1) Added. Will make another release in a few days

Feel free to do a PR if find anything else

jonathan-schmieg, 2023-01-20:
@jdeath don't want to open a new issue just to say thanks again!
I have implemented the external component like you suggested in the readme
it works! thank you