jdeath/rd200v2 #7: Issue with integration: Retrying setup: Unable to fetch data: unpack requires a buffer of 2 byetes

tvories, 2023-01-07:
The integration has recognized my sensor, but I get these errors when trying to initialize.

[image]
[image]
[image]
[image]

jonathan-schmieg, 2023-01-07:
@tvories try restarting home assistant - I get a single reading when doing that. Details in the issue I opened.
@jdeath are you OK with me closing the other issue and continuing in here? Seems to be the same thing and my original problem is solved

jdeath, 2023-01-07:
There seems to be some stability issues. Mine has been 100% stable...until yesterday. Other users have reported it too. I think the Bluetooth stack is just unstable and still a work in progress. Most other users have it stable for a few hours, then it drops. That is what I am seeing now. I am not really sure what to do. As hack, people remove this line:    https://github.com/jdeath/rd200v2/blob/b4ff3ff0bd9bbfbc3d9a2e90e07392a48250b287/custom_components/rd200_ble/rd200_ble/parser.py#L151

You can keep both issues open for now.
and it seems to be more stable. It is a hack and not ideal, but it might work. I am also going to try rebooting the RD200, as that is the only thing that changed on my end.

jdeath, 2023-01-07:
I just pushed a new parser.py. Try manually downloading it (get the raw file) and see if any better. It should at least avoid the unpacking error. Maybe that will make it more stable, not sure. I think I'll need a professional to look at this, but it is 99% stolen from the Airthings code, so I am not sure what else to change

jdeath, 2023-01-07:
@tvories Also make sure if you have ble_client: section in your esphome, it does not have the RD200 mac address listed. Almost move it closer or further away.

tvories, 2023-01-07:
@jdeath I have updated HACS with your recent updates and the device connected successfully!  I'll monitor things and see how long it keeps connected, but thank you for your efforts!

jdeath, 2023-01-07:
Yeah!

closed 2023-01-07 by jdeath