jdeath/rd200v2 #13: Minor Issue with entity type

nvOC-Stubo, 2023-03-14:
Hello:

I installed v. 2.6 yesterday and am loving it. Thanks for all of your hard work in developing it. I have noted one minor issue with the implementation of the radon_uptime_string sensor entity in HA. When core starts, I get this warning in the log:

Sensor sensor.fr_gl05re******_radon_uptime_string has device class None, state class measurement and unit None thus indicating it has a numeric value; however, it has the non-numeric value: 8d 17:02:65 (<class 'str'>); Please update your configuration if your entity is manually configured, otherwise report it to the custom integration author.
4:55:45 AM – (WARNING) Sensor				  

Thank You!

jdeath, 2023-03-14:
weird. that does not show up for me. Just updated to latest home assistant too. Try changing line 78 in sensor.py to
`state_class=None,` and see if that removes the warning

nvOC-Stubo, 2023-03-14:
Making that change and restarting core made no difference. I also tried a reload of the integration and a manual entity update just as a sanity check. I dug a little deeper into core.entity_registry and see this for that entity:

        _"capabilities": {
          "state_class": "measurement"
        },_

I looked at some other entities and suppose I could stop core and change it to:

        _"capabilities": null,_

and that should fix it. I would prefer to do this the proper way, if possible, but am not sure what the proper way is. At least I have a fix that I know will work as a fallback.

jdeath, 2023-03-14:
Weird. All the settings to change in sensor.py are documented here: https://developers.home-assistant.io/docs/core/entity/sensor/
It says setting to None should treat it as non-numeric. Maybe add native_value=None, to the list . Good luck. Post a solution if you find one

nvOC-Stubo, 2023-03-14:
I tried changing the core.entity_registry as above but HA just changed it back during startup. I also tried overriding the state_class in my customize yaml. The good news is that it works to set the state class as confirmed by examining the entity in Developer Tools|States. The not so good news is that I still see the error on startup.

I will post again if I find a solution that does not involve removing the deleting the entities and the integration and starting over with the change in sensor.py already in place.

jdeath, 2023-07-17:
closed with latest release.

closed 2023-07-17 by jdeath