jdeath/rd200v2 #16: radon_uptime_string wrong type

Din-BH, 2023-05-22:
Seems the type for the value of radon_uptime_string is not correct. Of course the issue goes away if you disable the sensor, but if one would like to keep the sensor, then a small fix would probably be needed. Log below.

```
2023-05-22 16:20:18.601 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform rd200_ble
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 579, in state
numerical_value = int(value)
ValueError: invalid literal for int() with base 10: '13d 03:53:02'
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 455, in async_add_entities
await asyncio.gather(*tasks)
File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 731, in _async_add_entity
await entity.add_to_platform_finish()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 845, in add_to_platform_finish
self.async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 585, in async_write_ha_state
self._async_write_ha_state()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 649, in _async_write_ha_state
state = self._stringify_state(available)
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 591, in _stringify_state
if (state := self.state) is None:
File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 583, in state
raise ValueError(
ValueError: Sensor sensor.fr_re22207111153_radon_uptime_string has device class None, state class measurement unit None and suggested precision None thus indicating it has a numeric value; however, it has the non-numeric value: 13d 03:53:02 (<class 'str'>)
```

jdeath, 2023-05-25:
Yes, I am aware of this, but I have no idea how to fix it. I have tried many combinations, but I think a text sensor is not allowed for some reason even though it works fine. I'll just have to wait until someone does a PR and suggests a fix.

Makr91, 2023-07-17:
I believe the issues is that only 3 device classes, and having any State Class set other than none, requires a numeric value.

I have made PR with this fix, however I am not sure if any of the other sensors would be affected by this change. For the most part they all seem to have the below which I believe is why my sensors are still working.

```
state_class=SensorStateClass.MEASUREMENT,
```

jdeath, 2023-07-17:
Close with latest release

closed 2023-07-17 by jdeath