jdeath/rd200v2 #18: The state class '' of this entity is not supported

spikeygg, 2023-08-15:
I've been using this integration for a while now and it seems that I get this message in the Developer Tools-->Statistics results after a while:

"The state class '' of this entity is not supported"

[image]

It happens on both my units and clicking the "FIX ISSUE" button displays this:
[image]

jdeath, 2023-08-19:
I have not seen that issue. The state class should be measurement. Are you running the latest release?

spikeygg, 2023-08-19:
Yes, at least I think I am. Mine says version 2.9 is installed. That's the latest I see in the HACS box:
[image]

When I look at the entities in the developer tools things it doesn't show any `state_class` detail:
[image]

jdeath, 2023-08-19:
Try version 3.0. Should be fixed now.

spikeygg, 2023-08-19:
That's great news! However, I just upgraded and restarted HA and it says this in the logs now:

Logger: homeassistant.setup
Source: setup.py:379
First occurred: 8:17:39 AM (2 occurrences)
Last logged: 8:17:56 AM

Unable to prepare setup for platform rd200_ble.sensor: Platform not found (cannot import name 'BATTERY_ENERGY_DISCHARGED_SENSOR' from 'custom_components.rd200_ble.const' (/config/custom_components/rd200_ble/const.py)).

jdeath, 2023-08-19:
Opps. I uploaded a sensor.py from another integration! Try version 3.1

spikeygg, 2023-08-19:
Awesome! Thanks jdeath, that was the trick! I think I'm in business.

closed 2023-08-19 by spikeygg