jdeath/rd200v2 #20: sensor loses regularly connection reporting "unavailable"

coolje, 2023-09-10:
I have just installed the RD200-Integration to my HomeAssistant. 
I'm using RadonEye RD200 that is called FR:R20:SN0114 with Firmware Version 1.2.4
I'm HomeAssistant on an Intel NUC Windows10 within a VMWare Virtual Machine Actual Core is 2023.9.1 
Direct Bluetooth connection between the NUC and the RadonEye Bloototh.
Problem:
The RD200 loses from time to time the connection and the System reports "unavaiable". 
Sometime after sometime 1 hour sometime half a day RD200 will respond again sendig actual Radon Values.
When the connection is lost it normally helps to restart the RD200 device.
Only one time it was necessary to restart the "Radon Eye" Sensor with unplugging the power cord to it. 

Is it possible to restart the RD200 Software automatically when no data is receiving since i want to use the data to control a fan in my cellar where the Radon Level exceeds 1000Bq/m^3 without ventillation within 3 or 4 days. 
(and sorry for my bad english)

jdeath, 2023-09-10:
There is a service to restart integrations: homeassistant.reload_config_entry , but I never used it: https://community.home-assistant.io/t/wth-is-there-no-reload-integration-service/467191/24

The BLE connection is brittle. First, I recommend getting a ESP32, making it a BLE proxy, and put it near your RD200 device.  https://esphome.github.io/bluetooth-proxies/ That will work *much* better than a built in Bluetooth on a NUC.

Also, may want to consider professional mediation versus using this as a trigger. I hope you mean 100 Bq/m3 because 1000 is crazy high!

coolje, 2023-09-10:
No 1000Bq/m^2 is correct. I can manage values ~50Bq/m^2 by with a fan, but without,  in 3 to 4 days it can raise to >1000Bq/m^2.  My ground is naturally Radon richt (natural Uranium) and the house is almoust 100 years old and therfore has not considered to be Radon leakproof. 
In germany 300Bq/m^2 where allowed for years but now the EU reduced the border to values less than 100Bq/m^2 on new buildings - old buildings do no
I do not want to spend time in ESP32 integration, another new matter. 
I will look in the

coolje, 2023-09-28:
I want to give a feedback.
I solved my "regularly loosing-connection" on my Win10/Nuc running Home-Assistant in a VMWare Virtual machine by updating from my licenced VMware Pro 15 to VMWare (free) Verions 16. The RD200v2 now works without any lost value since 14 days, so updating VMware helped. 
(unfortunately my expensive licence key for VMWare 15 is not valid for v.16, so i cannot run Home-Assist in VMware 16 (free) as as Windows 10 service as background without loging in a user (i failed in configure it manually as an service in win10), starting VMware as user - but this is a new problem not concernig this great Adon)

jdeath, 2023-10-07:
Thanks. I'll put this in the readme.

closed 2023-10-07 by jdeath