jdeath/rd200v2 #30: RD200 sensors work for an hour, then stop until BLE proxies or HA are rebooted

dimatx, 2023-12-04:
It's unlikely this is an issue with this integration, but wanted to see if others may have found a workaround.

The RD200 goes 'Unavailable' until BLE proxies or HA are rebooted. Other BLE sensors that are near and far continue to work though without requiring reboot.

I have a couple ESP32-based Bluetooth proxies. One is in close proximity next to the RD200.

The error I get in HA is:
`Error fetching rd200_ble data: Unable to fetch data: 94:E6:86:88:70:9A - 94:E6:86:88:70:9A: Failed to connect after 10 attempt(s): No backend with an available connection slot that can reach address 94:E6:86:88:70:9A was found: The proxy/adapter is out of connection slots or the device is no longer reachable; Add additional proxies (https://esphome.github.io/bluetooth-proxies/) near this device`

jdeath, 2023-12-04:
Yes, this is just how it works. Mine will usually just working again after a couple missed updates. Make sure you are running the latest ESPHome. Mine has been up for 76 days, but usually miss a reading a few times a day. The radon eye using a longer-than-standard notification data length, which I think causes issues.

dimatx, 2023-12-05:
Yeah, my issue is that it wasn't coming back for hours/days. I am not sure if I've fixed it, but by disabling sensors I didn't need (like the pulse ones), it seems to be working now without going unavailable. Possibly a coincidence? Not sure. (Also, just as a data point, this wasn't an issue for me until I upgraded to the latest ESPHome earlier this month.)

All of my BLE adapters are proxies.

closed 2023-12-14 by dimatx