jdeath/rd200v2 #36: HK serial number prefix showing up !

brucesemple, 2024-01-15:
Since this custom integration is working so nicely with the latest releases of ESPHome and HA .. I ordered a second RD200 Unit.
It's reporting the same firmware version 3.01 on the screen (as my FR:HA unit)  .. but it looks like they have gone to the next prefix in their serial numbers .. I got a HK  (FR:HK) unit.  I'm in the US.   I see your instructions on what source files need to be updated, however I don't know how to load an updated version of this integration into HA for a test.

jdeath, 2024-01-15:
Try latest release. Should support FR:HK

brucesemple, 2024-01-15:
OH.. silly me .. I did not realize these files were in the /config directory ..   Nice job with the integration .. both devices are showing up now.  tnx

closed 2024-06-28 by jdeath