jdeath/rd200v2 #44: HACS custom integration installation not working with 2024.7.1?

ryan-roberts, 2024-07-08:
Running HA Core 2024.7.1. 

I added repository "https://github.com/jdeath/rd200v2" to HACS as a custom integration and restarted HA (and waited) but it doesn't show up like other HACS integrations.  I also tried trying to add the integration via the project's readme (Settings > Devices & services > Add Integration > Search "rd200") but that doesn't show anything either.

Trying to re-add custom repository results in message "_Repository 'jdeath/rd200v2' exists in the store._" yet it's not showing up in the Custom repositories HACS dialog. I can confirm it's listed in /homeassistant/.storage/hacs.repositories though (which is why that 'exists in the store' message is occurring).

jdeath, 2024-07-09:
That is strange. I would remove the entry from the .storage and restart. If added correctly, you should not need to restart HACS to see the added repository.

ryan-roberts, 2024-07-09:
Yes it is all strange-- if this is just a me problem then good news for everyone else but I wanted to drop a comment just in case it wasn't. 

I did try removing the entry in hacs.repositories to retry the whole thing but the process played out in the same way. I even tried manually copying the custom_components files into /homeassistant/custom_components but it wasn't picked up on restart (and I didn't debug any further).

jdeath, 2024-07-09:
That sounds weird. You place the entire rd200v2 folder into \\homeassistantIP\config\custom_components\` . You can then go to Settings->integrations and add an rd200 integration, but it should just find the device and ask you if you want to set it up after you restart. 

I have not had reports of this. But will leave the issue open, just in case others chime in.

ryan-roberts, 2024-07-20:
Thanks for the step clarification. I had copied rd200_ble into /homeassistant/custom_components/rd200_ble but didn't add the integration. I've now got the integration working, thank you! 
[image]

closed 2024-07-20 by ryan-roberts