jdeath/rd200v2 #45: Unable to fetch data: Service Discovery has not been performed yet

Morac2, 2024-10-14:
I set this up with HomeAssistant which is installed as a docker component on a Raspberry Pi 5b for a newly bought RadonEye.

I disabled checking for uptime and peak, yet the check still frequently fails with the following error. 

“Error fetching rd200_ble data: Unable to fetch data: Service Discovery has not been performed yet”

When that happens the valise all change to unavailable until the next successful check is made. The success rate seems to be maybe 25% or less.

I have 4 other BTLE devices which seem to work fine or at least they aren’t generating similar errors.

Morac2, 2024-10-14:
I tested out this project https://github.com/sormy/radoneye which works on my MacBook Air, but gives the same service discovery error about 75% of the time on my RP5b, so the problem is with the Pi.  On a side note, that project can do a lot more such as getting the history.

It could probably be worked around by retrying on a failure, though I'm not sure if that's possible or not.

jdeath, 2024-10-14:
Feel free to edit the code and do a PR to add a retry capability. Most BLE integrations just wait to the next update. The proxy will make it much better.

Morac2, 2024-10-25:
I got a couple of Atom lite devices set up as ESPHome proxies and the errors are almost completely gone now.  It’s working properly.