jdeath/rd200v2 #46: Error fetching rd200_ble data: Unable to fetch data: 'NoneType' object has no attribute 'address'

CodeFinder2, 2024-12-14:
This seem to occur once after startup (restarting HA). Not sure where this is coming from.

Using an FR:HJ device (paired successfully out of the box). Do you need more info?

jdeath, 2024-12-14:
Sometimes the update just fails. It should work on a subsequent update if you wait long enough. Are you using a Raspberry PI bluetooth, those are buggy? Esphome proxies work best.

CodeFinder2, 2024-12-14:
Ah sorry, forgot to mention, I am using an x86 thin client with an Intel Wifi/Bluetooth card, being passed to a HA VM running in Proxmox.

Apart from that error message, updates seems to work fine.

If it's fine, shall we convert that error to a warning or just an info?

jdeath, 2024-12-14:
Info is probably fine. I do not know how to catch the error and throw an info error. Please do a PR if you have a solution.

CodeFinder2, 2024-12-19:
Will take a look once it happens again. I am currently unable to reproduce it.

closed 2024-12-19 by CodeFinder2