jdeath/rd200v2 #49: Upgrading to 4.8 broke support for my I series

mcattle, 2025-07-04:
Just a heads-up, upgrading to version 4.8 (J series support) broke support for my I series detector, but reverting back to 4.7 got it working again.

[image]

jdeath, 2025-07-05:
Hmm...not sure why. the release only added support for J. Do not see why I would not be supported, the code looks ok.

anyways, 4.8 does not add anything else so just stay at 4.7. Really, never any need to update unless Home Assistant depreciates a constant (which they like to do every year or so without telling anyone!)

florian7393-dotcom, 2025-11-12:
Hey guys,

im completly new into HA and stuff, but i got my Bluetooth and stuff running on my HA but i get it working... i guess its because of the SN... JE01RE0008800. Manufactured in Jul. 2025. Can anybody Help me?

florian7393-dotcom, 2025-11-14:
Nevermind, installed an EDS32 Mini today and deleted the other Bluetooth devices. and Re installed this repository (hope thats right) and now i could find it.  *HYPE*